Labsmore/pyuscope #30: Use TOML for configuration

FFY00, 2020-09-08:
TOML IMO provides a much nicer configuration language than JSON, which would help make this project just a little bit more user friendly.

TOML mappings are really close to JSON mappings, I am pretty sure we don't need to any custom logic. The API is the same, we should be able to just use `toml` module instead of `json` for TOML configs.

https://github.com/toml-lang/toml

https://github.com/uiri/toml#api-reference

This is just a sugestion :blush:

JohnDMcMaster, 2020-09-08:
Will take a look. Priority right now though will probably go to other areas

JohnDMcMaster, 2022-08-31:
Thanks for the feedback but after careful consideration we've decided to move to JSON5. This was mostly due to JSON to JSON5 being a trivial transition to fix our pain points but its was less clear if TOML would offer any significant advantages over JSON5 for our use case.

closed 2022-08-31 by JohnDMcMaster