Labsmore/pyuscope #48: lip-a1 software stack install steps

yusufm, 2022-08-22:
- Ubuntu 20
- sudo apt install git
- git clone https://github.com/labsmore/pyuscope.git
- copy over lip-a1 config dir to configs
- Follow instructions for camera drivers at: https://github.com/JohnDMcMaster/gst-plugin-toupcam
- sudo apt install pip3
- pip3 install setuptools
- sudo python3 setup.py develop
- sudo apt-get install -y python3-gst-1.0 python3-gi python3-pyqt5 python3-usb
- pip3 install pyserial
- pip3 install paramiko
- sudo apt install. libxcb-xinerama0

- ./RUN_me.sh

JohnDMcMaster, 2022-08-31:
I made a first cut at w/ install_lip-a1_ubuntu_20.04.sh but I think its missing some of the things above (ex: paramiko)

yusufm, 2022-08-31:
do a pip freeze and then add the requirements file. that would work for
now.

JohnDMcMaster, 2022-09-06:
```
bash configs/lip-a1/setup_ubuntu_20.04.sh
```

this was done, although I don't think I have sudo apt install. libxcb-xinerama0 . Where did you encounter that?

JohnDMcMaster, 2023-01-07:
seems this is as done as it will be. there is a separate ticket for general setup, add more info there if needed

closed 2023-01-07 by JohnDMcMaster