Labsmore/pyuscope #116: Increase video widget size

JohnDMcMaster, 2023-03-17:
Why so small

travisgoodspeed, 2023-04-07:
A jerry rigged fix to `gstwidget.py` that makes the software functional on my 4K monitor:

```python
    def set_gst_widget_wh(self, wigdata, w, h):
        assert wigdata["capsfilter"] is None, "FIXME: handle gst initialized"
        w = int(w)
        h = int(h)
        w = 1500    #Ignore the real value for a forced replacement.                                                           
        h = 1200
        wigdata["width"] = int(w)
        wigdata["height"] = int(h)
        # might not be initialized yet                                                                                         
        if wigdata["widget"]:
            wigdata["widget"].setMinimumSize(w, h)
            wigdata["widget"].resize(w, h)
```

I can't merge that, of course, so I'll doodle around with this and then send a proper pull request.

JohnDMcMaster, 2023-04-07:
Yeah was thinking a good short term fix might be to shoehorn a config option to let users manually tweak

JohnDMcMaster, 2023-05-24:
Working on overhauling the video rendering calculations right now. This should allow me to fix this and also maybe a floating full screen widget

JohnDMcMaster, 2023-05-26:
Major overhaul to this system. Changes:
- Widget resizing code should be a lot better now
- You can create a second window with a screen sized video feed using thew new Video menu at the top and selecting "full show"
- General code cleanup

Notes / known issues:
- If you have multiple monitors of different resolutions this might behave unexpected
- Under-zooming ROI will complain loudly and give a black widget
- Consider making the full and ROI widget the same size to make more symmetrical

Other: was also a bit of a tech demo including:
- Dynamically changing pipeline properties (video menu can also zoom in/out of ROI)
- Dynamically adding elements to the video pipeline

closed 2023-05-26 by JohnDMcMaster