Labsmore/pyuscope #148: Add joystick support

yusufm, 2023-04-07:
Joystick support would greatly help certain user flows in pyuscope. 

One of the suggested joystick requested is https://www.amazon.com/Logitech-Joystick-Programmable-Weighted-Rapid-fire/dp/B00009OY9U/ref=asc_df_B00009OY9U/

And the possible key mapping for the above joystick is:

Joystick X/Y = stage X/Y

Trigger = snap photo

Throttle = sensitivity, up = move faster / down = move slower

Hat up / down = adjust focus

There are other modes (like move and mark corners for setting scan bounding box marks) that would benefit and need consideration on the key mappings. 

Note that having a configurable setup to allow for different controllers (e.g. Xbox controller) would be beneficial.

yusufm, 2023-04-07:
Should also potentially have a checkbox in the UI to achieve I ate the joystick.

And we should disable joystick during a scan.

JohnDMcMaster, 2023-04-19:
Seems like basic version is working. Closing ticket

closed 2023-04-19 by JohnDMcMaster