Labsmore/pyuscope #159: auto update

yusufm, 2023-04-18:
Would be nice if pyuscope had a check for / install update feature.

Interim solution might be on startup we check for a git hash we wrote out, and run the install script if it doesn't match.

JohnDMcMaster, 2023-04-18:
I'd be up for notifying if a new tagged release is out. But anything, especially an auto-update, would need to be well thought out. I wouldn't suggest that right now

FFY00, 2023-07-17:
As someone intimately involved in Python packaging ecosystem, I would strongly advise against that. Just spend that time making the software better 😅

yusufm, 2023-07-17:
@FFY00 we would love for people to help make the software better ;)   Send in those PRs :)

JohnDMcMaster, 2023-07-18:
@yusufm we should consider a CLA before taking (more) PRs. I'm happy to not funnel PRs for now