Labsmore/pyuscope #166: GstVideoTestSrc does not have property "serial-number"

duckythescientist, 2023-05-07:
Looks like the `GstVideoTestSrc` is missing the `serial-number` attribute. I've patched [plugin.py](https://github.com/Labsmore/pyuscope/blob/5dca3db85e6ebe3b46c78bbb619689d7d3f4f2d2/uscope/gui/plugin.py#L49) `get_sn` locally to `except TypeError` and return `None`, but I'm guessing there's a better or lower level way to fix this. I'm still familiarizing myself with the codebase. 

`gi.__version__ == '3.36.0'`

```
Traceback (most recent call last):
  File "/home/duck/Software/pyuscope/uscope/app/argus/main.py", line 200, in main
    gstwidget_main(MainWindow, parse_args=parse_args)
  File "/home/duck/Software/pyuscope/uscope/gui/gstwidget.py", line 622, in gstwidget_main
    _mainwin = AQMainWindow(**kwargs)
  File "/home/duck/Software/pyuscope/uscope/app/argus/main.py", line 37, in __init__
    self.post_ui_init()
  File "/home/duck/Software/pyuscope/uscope/app/argus/main.py", line 150, in post_ui_init
    self.ac.post_ui_init()
  File "/home/duck/Software/pyuscope/uscope/app/argus/common.py", line 213, in post_ui_init
    self.microscope.configure()
  File "/home/duck/Software/pyuscope/uscope/microscope.py", line 89, in configure
    configure_motion_hal(self)
  File "/home/duck/Software/pyuscope/uscope/motion/plugins.py", line 96, in configure_motion_hal
    "scalars": usc.get_motion_scalars(microscope),
  File "/home/duck/Software/pyuscope/uscope/config.py", line 765, in get_motion_scalars
    system = self.find_system(microscope)
  File "/home/duck/Software/pyuscope/uscope/config.py", line 638, in find_system
    camera_sn = microscope.imager.get_sn()
  File "/home/duck/Software/pyuscope/uscope/gui/plugin.py", line 49, in get_sn
    return self.ac.control_scroll.raw_prop_read("serial-number")
  File "/home/duck/Software/pyuscope/uscope/gui/control_scroll.py", line 498, in raw_prop_read
    return source.get_property(name)
TypeError: object of type `GstVideoTestSrc' does not have property `serial-number'
```

JohnDMcMaster, 2023-05-07:
The submodule gst-plugin-toupcamsrc reference needs to be updated. Can you manually update to the latest?

duckythescientist, 2023-05-07:
Didn't seem to make a difference. 

Doesn't `gst-plugin-toupcamsrc` only install the toupcam plugin? `GstVideoTestSrc` is part of GStreamer proper.

JohnDMcMaster, 2023-05-08:
Oh I see. Yeah that's a different issue.

JohnDMcMaster, 2023-05-13:
Fixed!

closed 2023-05-13 by JohnDMcMaster