Labsmore/pyuscope #167: The `mock` configuration is broken (missing "systems")

duckythescientist, 2023-05-07:
After patching #166, `./app/argus.py --microscope mock` "failed to either match system or find default for camera S/N None". I think the mock j5 file needs to be updated to have a "systems" section. There are probably other configs that are outdated. (And as I mentioned in #166, I'm still  familiarizing myself with the codebase. Hopefully before too long I can submit PRs instead of just issues.)

A version of `microscope.j5` that works:

```
/*
A minimal configuration to do basic testing
*/
{
    "motion": {
        "hal": "mock"
    },
    "imager": {
        "source":"gst-videotestsrc",
        "width": 150,
        "height": 50
    },
    "systems": [
        {
            "camera_sn": null,
            "objectives": [
                {
                    "name":"Objective",
                    "suffix":"suffix",
                    "x_view": 100.0
                }
            ]
        }
    ]
}
```

JohnDMcMaster, 2023-05-07:
Ah yeah this hasn't been tested recently. I'll give it some TLC

JohnDMcMaster, 2023-05-13:
Fixed!

closed 2023-05-13 by JohnDMcMaster