Labsmore/pyuscope #244: lip-vm1b crashes after scan

yusufm, 2023-08-19:
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
buf 960 960 1843200
failed on White Balance Temperature 4600
<class 'OSError'>: [Errno 84] Invalid or incomplete multibyte or wide character
  File "/home/labsmore/pyuscope/uscope/app/argus/widgets.py", line 1033, in plannerDone
    self.ac.imager.set_properties(self.restore_properties)
  File "/home/labsmore/pyuscope/uscope/imager/imager.py", line 80, in set_properties
    self._set_properties(vals)
  File "/home/labsmore/pyuscope/uscope/gui/plugin.py", line 94, in _set_properties
    self.ac.control_scroll.set_disp_properties(vals)
  File "/home/labsmore/pyuscope/uscope/gui/control_scroll.py", line 324, in set_disp_properties
    self.prop_write(element.config["prop_name"], val)
  File "/home/labsmore/pyuscope/uscope/gui/control_scroll.py", line 384, in prop_write
    self.raw_prop_write(name, value)
  File "/home/labsmore/pyuscope/uscope/gui/v4l_control_scroll.py", line 108, in raw_prop_write
    v4l2_util.ctrl_set(fd, name, val)
  File "/home/labsmore/pyuscope/uscope/v4l2_util.py", line 142, in ctrl_set
    fcntl.ioctl(fd, v4l2.VIDIOC_S_CTRL, control)

JohnDMcMaster, 2023-08-19:
Aside:

```
buf 960 960 1843200
```

these are an old debug statement, we can remove them

JohnDMcMaster, 2023-08-29:
For what its worth I still haven't seen this, but we can keep an eye out for it

JohnDMcMaster, 2023-09-17:
this was fixed by moving to new v4l2 api

closed 2023-09-17 by JohnDMcMaster