Labsmore/pyuscope #270: VM1b camera error (v4l2.VIDIOC_S_CTRL)

yusufm, 2023-09-04:
The following error started occurring at https://github.com/Labsmore/pyuscope/commit/a03a1a2be26d32a5b61c3a9931144d268e7a7ae7

```
$ ./app/argus.py --microscope lip-vm1b
pygame 2.5.1 (SDL 2.28.2, Python 3.8.10)
Hello from the pygame community. https://www.pygame.org/contribute.html
Camera 'Hayear Camera: HY-800B': selected /dev/video0
GRBL homing w/ hack, selected tool: 0
Sending home command 1
Homing successful after 41.9 sec. Ready to use!
Joystick: detected Logitech Extreme 3D pro
failed on Exposure Time, Absolute 157
<class 'PermissionError'>: [Errno 13] Permission denied
  File "/home/labsmore/pyuscope/uscope/gui/control_scroll.py", line 149, in gui_changed
    self.cs.raw_prop_write(self.config["prop_name"], val)
  File "/home/labsmore/pyuscope/uscope/gui/v4l_control_scroll.py", line 109, in raw_prop_write
    v4l2_util.ctrl_set(fd, name, val)
  File "/home/labsmore/pyuscope/uscope/v4l2_util.py", line 142, in ctrl_set
    fcntl.ioctl(fd, v4l2.VIDIOC_S_CTRL, control)
```

JohnDMcMaster, 2023-09-04:
This probably won't get resolved without some access to your system that is showing an issue. However, I see you are using a joystick. Does this happen when you don't use a joystick?

yusufm, 2023-09-04:
Yes, happens without joystick as well.

JohnDMcMaster, 2023-09-08:
I got a laptop here now that reproduces the issues. Observations:
- Occurs in both wayland and cinnamon
- Seems to happen to every other control? If I skip a control the next one is problematic
- File descriptor seems constant / not corrupted
- I tried allocating the query objects a little differently and didn't seem to make a difference
- I tried disabling the non-base ranges and still got issues
- Repeating the call on failure does not help

Next steps:
- Hexdump the query structure?
- More google searches
- Can we make a reduced test case?

JohnDMcMaster, 2023-09-13:
I've made a quick attempt to port the existing code to use a new API (https://github.com/antmicro/pyrav4l2). However they don't have any examples read / writing controls and I'm getting some unexpected output back from get_control_value() and I'm unclear if I'm using the API wrong or if this is a bug. I'll dig a bit deeper

JohnDMcMaster, 2023-09-14:
fixed by switching to above framework

closed 2023-09-14 by JohnDMcMaster