Labsmore/pyuscope #308: Distribution (packaging)

FFY00, 2023-10-11:
I am opening this issue to figure out the software distribution user story. We should figure out what is our desired UX within the limitations of the packaging tooling, software architecture, platform requirements, etc.

#### Distribution format

Options, and key considerations:

- Self-contained application
  - Easy to use
    - User gets a single file and just needs to run it
  - Large size
  - Unable to access system GStreamer plugins
- Via `pip install`, or equivalent
  - Possibly difficult to use
    - User needs to install the Python package, which can fail for multiple reasons
  - Useful for power-users
- Distribution (Debian, Ubuntu, etc.) specific packages
  - Easy to use
  - Limited environment/platform support
    - It's not feasible to make packages for all possible systems
  - Needs a significant ongoing effort to keep packages up-to-date
- Other environment management solution (snap, flatpak, etc.)
  - Mixed difficulty to use
    - If the user already has snap, flatpak, etc. installed, they just need to install the application
    - If the user does not have snap, flatpak, etc. installed, they need to install it, and go through the initial learning curve
  - Hardware access could be problematic
  - Unable to access system GStreamer plugins

Extra considerations:

- Self-contained application
  - Plugin feature needs a custom solution
  - The extra scripts that we currently have need to be bundled into a single CLI (eg. `pyuscope script foo --somearg`)
- Via `pip install`, or equivalent
  - Can be used as a base for other kind of installations
- Other environment management solution (snap, flatpak, etc.)
  - Plugin feature needs a custom solution

#### Dependencies

The main thing to consider when gauging the viability of a certain distribution method are the dependencies. I tried to document them here.

- Python
- Pure-Python 3rd party packages
  - `boto3` (hard dependency, `cloud_stitch`)
  - `json5` (hard dependency, `joystick`)
  - `pexpect` (hard dependency, `motion.lcnc.rsh`)
  - `paramiko` (optional, `hal_ar`)
  - `pyusb` (optional, `imager.imager_util`)
  - `v4l2` or `pyrav4l2` (optional, for V4L2 support)
  - `imageio` (script dependency, `utils/zoom_gif`)
  - `linuxcnc` (remote server dependency)
- Native 3rd party packages
  - `numpy` (hard dependency, `imagep`, `utils/zoom_gif`)
  - `opencv-python` (hard dependency, `cal_util`, `imagep`, `imager.autofocus`)
  - `pillow` (hard dependency)
  - `PyGObject` (hard dependency)
  - `PyQt5` (hard dependency, `argus`)
  - `scipy` (hard dependency, `planner.plugins`)
  - GStreamer (hard dependency)

We will always depend on Python, so that's our biggest limitation, followed by the dependencies on native 3rd party packages. Still, I think it should be possible to use a tool like [PyInstaller](https://pyinstaller.org/en/stable/), making the self-contained application option viable, the native dependencies should be the limitation here.

Depending on native 3rd party packages makes the `pip install` option a bit more problematic, as our application will pull native packages, which increases the complexity and possibility for failures. The UX here would largely be dictated by how the native dependencies are packaged. In our case, all our native dependencies are fairly big projects and seem to be well packaged — they provide binary packages (also called *wheels*) for Linux, macOS, and Windows, and either use the [stable ABI](https://docs.python.org/3/c-api/stable.html) (`abi3`), or provide binary packages for a significant set of Python versions.

#### Proposed resolution

(note: this is just my opinion, after considering the points documented above)

IMO, we should package the app for the Python packaging ecosystem, and make it available on [PyPI](https://pypi.org), which would allow users to `pip install pyuscope`. This would mostly target power-users, but also provide a build system for packaging to most other formats (eg. [see the Arch Linux packaging file for python-libusb1](https://gitlab.archlinux.org/archlinux/packaging/packages/python-libusb1/-/blob/main/PKGBUILD?ref_type=heads)).

To target normal end-users, I would try to get the app packaged into a self-contained file using [PyInstaller](https://pyinstaller.org/en/stable/).

FFY00, 2023-10-11:
Regarding packaging for PyPI, we are already using setuptools, so we are almost there, the only thing missing to be able to build distributable Python packages is to specify the dependencies in `setup.py`.

JohnDMcMaster, 2023-10-11:
Hi there! Thanks, looked over this briefly and a few quick notes:
- There is a proprietary binary blob used for touptek cameras (ex: X1, LS-HVY microscopes). How / can we distribute that?
- RPI support keeps coming up as desirable, so we should ideally make something that has a roadmap towards that
- LinuxCNC/machinekit is generally deprecated and I don't see it likely to come back on our roadmap. As of 2023 all of my machinekit systems have been converted to GRBL. So I wouldn't worry about packaging anything up related to that. I've considered ripping it out entirely

JohnDMcMaster, 2023-10-11:
also regarding hard vs optional dependencies: my preference for anything we are shipping in production support (say joystick is production supported vs linuxcnc which is deprecated) we should make a hard requirement unless we have a good reason to do so. While making some things optional made since at the time, now it might just be creating additional configurations we have to test

FFY00, 2023-10-11:
> There is a proprietary binary blob used for touptek cameras (ex: X1, LS-HVY microscopes). How / can we distribute that?

What is the license for that blob? As long as it allows it to be redistributed, we should be fine.

So that we have a good separation, I'd probably put binary blobs in a separate Python package (eg. `pyuscope-vendor-blobs`). The blobs can be installed as part of a package module, which we can then access with [import.resources](https://docs.python.org/3/library/importlib.resources.html).

Eg. Accessing a blob named `touptek-ABC.bin` in the `pyuscope.blobs` package.

```
$ tree
.
└── pyuscope
    ├── blah_blah_blah.py
    ├── blobs
    │   └── touptek-ABC.bin
    ├── __init__.py
    └── my_code.py

3 directories, 4 files
```

```python
>>> import importlib.resources
>>> importlib.resources.files('pyuscope.blobs') / 'touptek-ABC.bin'
PosixPath('/home/anubis/test/pyuscope/blobs/touptek-ABC.bin')
```

Note: A [importlib_resources](https://github.com/python/importlib_resources) backport package is available, for older Python versions.

> RPI support keeps coming up as desirable, so we should ideally make something that has a roadmap towards that

That sounds good, and is compatible with all the mentioned options. I'd target that either using the main approach we take for distribution, which IMO should be the self-contained app, if viable, but providing a RPI-specific package would also be an option here.

> also regarding hard vs optional dependencies: my preference for anything we are shipping in production support (say joystick is production supported vs linuxcnc which is deprecated) we should make a [...]

Yeah, that seems like the best approach to me. Pure-Python dependencies are cheap, and it's generally better to just pull them anyway. Native dependencies are more problematic, so I'd gauge the benefits there and decide based on that. For native dependencies, I'd also consider if they are a popular package (eg. `numpy`, `scipy`, etc), which I think should be less problematic, and should have more information available in case anything goes wrong.

JohnDMcMaster, 2023-10-18:
Here is the official download for the binary blob: http://www.touptek.com/download/showdownload.php?lang=en&id=32

At a quick glance I can't find any redistribution or copyright information with any meat to it. Ex: the code header doesn't have any such info nor can I find a EULA. This may be in part related to them doing weird rebranding things which could have made it less appealing to put copyright strings around the code.

JohnDMcMaster, 2026-10-23:
significant improvements here. let me know if you have more thoughts

closed 2026-10-23 by JohnDMcMaster