Labsmore/pyuscope #309: Modernize the Python packaging definition

FFY00, 2023-10-11:
Modern packaging tooling has moved towards the use of static files for the package definition.

I propose replacing the current `setup.py` file with a `pyproject.toml`. See https://setuptools.pypa.io/en/latest/userguide/pyproject_config.html

We can then also consider adopting a build backend that makes the development workflow nicer, like [flit](https://flit.pypa.io/en/stable/), [hatch](https://hatch.pypa.io/latest/), or [pdm](https://pdm.fming.dev/latest/), but staying with `setuptools` is still perfectly reasonable.

#### Complications

We have the following code in `setup.py`, which doesn't seem to have any effect right now, as there are no `util` or `demo` directories in the root of the repo. I image that was a mistake, though, and we intend on picking up `utils`, and maybe `app`.

https://github.com/Labsmore/pyuscope/blob/cee6789ae1a95b70d8fe70e611a1bc55ebfb4c13/setup.py#L17-L40

While we can still install scripts to `PATH`, it needs to be done slightly differently, but I'd suggest to simply move these scripts to a subpackage, making then callable via the Python CLI, eg. `python -m pyuscope.utils.microscope_info`. Then, if necessary, we can also put them in `PATH`, using [entrypoints](https://setuptools.pypa.io/en/latest/userguide/entry_point.html). I'd avoid cluttering `PATH` too much, though, and also make their provenance more discoverable, by prefixing them with `pyuscope-` for eg.

Another possible option would be to unify them into a single CLI, eg. `pyuscope-utils`, `pyuscope-demo`, etc.

---

#### Additional materials

`pyproject.toml` package definition overview and workflow: https://meson-python.readthedocs.io/en/latest/tutorials/introduction.html#configuring-our-python-package
`pyproject.toml` package metadata specification: https://packaging.python.org/en/latest/specifications/declaring-project-metadata/

FFY00, 2023-10-11:
I can send a PR for this, just let me know what your thoughts are.

JohnDMcMaster, 2023-10-18:
"[flit](https://flit.pypa.io/en/stable/), [hatch](https://hatch.pypa.io/latest/), or [pdm](https://pdm.fming.dev/latest/)" => there is a lot to unpack here. if you have a suggestion on the right direction we might just roll with it as my general feeling is that each thing we saw had advantages / disadvantages. With that in mind its probably better to make any decision rather than make the "right" / optimized decision

re: setup.py issues, I think that's just bitrot since this hasn't been looked at for a while as setup.py is not super useful / supported in its current form. However now that VA1/VM1 can be "pip install" (unlike X1) it might be more useful. The pyi page is quite out of date for example: https://pypi.org/project/pyuscope/

I'm not sure if I have strong opinions about where the installed files go, but it does need to be closer to "pyuscope-argus" rather than "python -m pyuscope.app.argus" which isn't discoverable / intuitive.

Related we haven't figured out what the "official" shipping set of apps should be. At a minimum its app/argus.py, but we probably also want at least utils/microscope_info.py. I'm not sure if there is a good reason why there is a "util" vs "app" directory and we might just merge those directories together. However, if something is under "test" we probably don't want to package it

JohnDMcMaster, 2026-10-23:
this was done recently

closed 2026-10-23 by JohnDMcMaster