Labsmore/pyuscope #314: autofocus does not respect soft limits

yusufm, 2023-10-18:
From Andrew using (4.2.0?):

if you're up near the top of the z axis range and hit AF it will mash into the limit switch

i.e. the soft limits are not respected during autofocusing

JohnDMcMaster, 2023-10-25:
Hmm I can't replicate this, at least on main: get "expected" behavior that the move is dropped with a warning. 

```
Selected objective M Plan Apo 20X
autofocus: coarse
autofocus round 1 / 7: try 1.018000
axis z: move violates -65.000 <= new pos 1.018 <= 1.000
autofocus round 2 / 7: try 1.012000
axis z: move violates -65.000 <= new pos 1.012 <= 1.000
autofocus round 3 / 7: try 1.006000
axis z: move violates -65.000 <= new pos 1.006 <= 1.000
autofocus round 4 / 7: try 1.000000
autofocus round 5 / 7: try 0.994000
...
```

While not ideal, this doesn't result in a limit switch crash like you are reporting.

Are you able to give me the final z position at failure? If the GUI crashes you can get (in machine coordinates) by running "python3 test/grbl/status.py " For reference, pyuscope Z coordinate +1.0 is GRBL coordinate -32.000

There could be some other explanations too like that we are simply too close to the limit switch now or the Z axis has drifted. But lets try to collect above first in case its an issue on your system

Other: there have been significant soft limit updates, but I think you updated past the critical commit. Can you please provide the commit you are on?

JohnDMcMaster, 2023-12-03:
Closing for now until we can get better reproduction info

closed 2023-12-03 by JohnDMcMaster