Labsmore/pyuscope #362: Add support for Raspberry Pi Cameras via picam2

philpem, 2023-12-14:
This ticket is a request for comments and collection of the discussions I've had on the Discord about this little project of mine.

I've been building an XYZ scanning microscope, using a Raspberry Pi 4 and [High Quality Camera](https://www.raspberrypi.com/products/raspberry-pi-high-quality-camera/) mated to an Olympus BHM (BH2-era) using an Olympus C-mount adapter on the trinocular port. Motion control is with a RAMPS1.4 board running GRBL and some motors I liberated from a broken 3D printer.

My initial plan was to modify the Pyuscope gstreamer backend to support the Libcamera source, but this had some issues:

- Setting the image resolution to the full resolution of the camera (4056x3040) corrupts the images coming out of the source. (the maximum resolution seems to be around 1080P, 1920x1080 pixels)
- No support for the Raspberry Pi Camera's settings - like white balance and exposure control
- The image format from the Gstreamer engine isn't what Pyuscope is expecting, causing a crash

I had a look at the [Libcamera Bugzilla](https://bugs.libcamera.org/buglist.cgi?component=gstreamer&product=libcamera) and it seems that the bugs above are known; the corruption issue may be [this buffer alignment one](https://bugs.libcamera.org/show_bug.cgi?id=169), which has seen no activity since it was reported in November 2022.
In regards to properties, the [GST camera provider](https://git.libcamera.org/libcamera/libcamera.git/tree/src/gstreamer/gstlibcameraprovider.cpp) only supports the Autofocus property. Nothing else is passed along to the camera.

I have no experience writing Gstreamer sources, so instead I opted to modify Pyuscope to add support for the Raspberry Pi project's "picam2" library. This turned out to be about a day's work to get something rough working: https://github.com/philpem/pyuscope-rpicam/tree/philpem_olympusbh_raspi

I'm open to comments on what I've done on this branch - I'm sure there are better ways to do some of the things I've done. It would be nice to get things into a state where they could be merged in -- though that's not a red line. If picam2 support isn't inline with the project goals then I'm OK with maintaining my fork on an 'it works for me' basis.

I've tested Autofocus, Snapshot and Panorama Capture and all seems to work fine.
Support for camera parameters/settings are missing - so everything's set to the defaults. I'm planning to add support for White Balance auto/mode/lock and Exposure auto/lock as a minimum, but will likely add some of the other Pi Camera features too.

JohnDMcMaster, 2023-12-16:
Hi philpem! Thanks for your info. Let me start by adding some info on my Supercon experiment with Mithro

Notes:
- We were testing misc RPI cams, although they were "standard" ones, not the HQ version
- For our demo we ultimately settled on libcamerasrc: https://github.com/Labsmore/pyuscope-kitchen/blob/main/configs/of-test/microscope.j5
- gst-plugin-libcamerasrc doesn't allow changing camera properties at runtime. However, you can preload a config file (JSON XML or something) to set them once
-  libcamera might be more flexible / it might be possible to write a better plugin, but it still might require restarting video pipeline when changing properties
- We investigated rpicamsrc which on paper is amazing. However, it uses deprecated kernel APIs so its "moving against the grain" even if it works well now. Notably these APIs are disabled by default on new RPI images
- There are some rpicam to USB adapters, but they are UVC => not great calibration experience. If we could get a GenICam version though I'd be more excited

philpem, 2024-02-10:
I've been working on this some more, and have a pretty usable version of this on my branch.

Imaging and panorama capture work as intended (or at least intuitively enough that I can't tell if there's breakage). Imaging controls also work, these are fairly simplistic. Not having support for floating-point values in the UI makes things a bit hairier than I'd prefer, but the multiplier factor is an OK workaround.

Development of this has uncovered a couple of other minor bugs in Pyuscope, for which I've opened separate issues.

JohnDMcMaster, 2024-02-10:
Thanks for filing the issues! Trying to triage and take a look over things. The biggest issue I need to figure out is what can be merged that isn't a maintenance burden to keep the other stuff moving. In my ideal world for example the video sources become more plugin like than they are now so changes can be better isolated