Labsmore/pyuscope #400: Exception in planner thread with LIP-VM1

edgetriggered, 2024-01-07:
Large scans are dying on loaner laptop (labtop3) at 75% point, seemingly caused by CPU frequency throttling having an effect on meeting a task deadline with three second timeout. Smaller test jobs with a few images to stitch run to completion.

dmesg is filled with messages like the following:
```
[90759.513304] mce: CPU0: Core temperature above threshold, cpu clock throttled (total events = 545405)
[90759.513305] mce: CPU3: Package temperature above threshold, cpu clock throttled (total events = 651553)
[90759.513306] mce: CPU1: Package temperature above threshold, cpu clock throttled (total events = 651551)
[90759.513307] mce: CPU2: Core temperature above threshold, cpu clock throttled (total events = 545404)
[90759.513308] mce: CPU2: Package temperature above threshold, cpu clock throttled (total events = 651551)
[90759.513309] mce: CPU0: Package temperature above threshold, cpu clock throttled (total events = 651553)
[90759.517352] mce: CPU2: Core temperature/speed normal
[90759.517352] mce: CPU0: Core temperature/speed normal
[90759.517354] mce: CPU1: Package temperature/speed normal
[90759.517355] mce: CPU3: Package temperature/speed normal
[90759.517355] mce: CPU0: Package temperature/speed normal
[90759.517356] mce: CPU2: Package temperature/speed normal
```

[pyuscope.log](https://github.com/Labsmore/pyuscope/files/13852344/pyuscope.log)

JohnDMcMaster, 2024-01-07:
Thanks for your report!

1) We can swap out your laptop. That might be the easiest fix here

2) We'll think over potential fixes here. Maybe a config directive that extends timeouts

JohnDMcMaster, 2024-01-07:
I've added a new config directive you can add to .pyuscope. Sample entry that increases timeouts from 3 seconds to 30 seconds. Can you pull latest code and let me know if this solves your issue?

```
{
       "timeout_scalar": 10.0,
       "labsmore_stitch": {
...
}
```

JohnDMcMaster, 2024-01-10:
Is this possibly related to the memory leak? That should now be fixed: https://github.com/Labsmore/pyuscope/issues/375

closed 2024-03-14 by edgetriggered