Labsmore/pyuscope #429: "Warning you have automatic exposure or color correction..." warning always shows False for colour correction state

philpem, 2024-02-10:
This is a fairly silly bug... 

If you start a panorama capture with AE and/or colour correction enabled, then Pyuscope displays a warning message. Or at least that seems to be the intention. Thanks to this line of code, the colour-correction status is always detected as "False":

https://github.com/Labsmore/pyuscope/blob/153938afffa290cef577fdc860cbba1ba7989267/uscope/gui/imaging.py#L1612

The colour correction status is read from the imager on the line above, and checked a few lines below.

This line (setting `auto_color = False`) needs to be deleted for this block to work as the warning message indicates it should.

JohnDMcMaster, 2024-02-10:
Eh this was probably some test code that was not intended to make it into production. That said it TLDR doesn't affect any Labsmore shipping models, so it probably got forgotten about.

philpem, 2024-02-10:
As far as I can tell, it's just a cosmetic gremlin which mucks up the warning message -- unless auto white balance/colour correction is enabled with auto exposure off. In that case the warning won't appear and the resulting panorama will be useless.

JohnDMcMaster, 2024-02-10:
Yes its just a warning / cosmetic, but I agree that it should trigger as intended

JohnDMcMaster, 2024-02-10:
I think I remember now: I vaguely recall this wasn't getting used in any production setups but was false triggering on mock source or something like this. So this was a quick fix / hack to stabalize things for the release. Should have had a comment and/or bug filed