Labsmore/pyuscope #430: Feature: GRBL control of microscope illumination

philpem, 2024-02-10:
GRBL has support for 'spindles'. It would be nice to have support for this, for switching the microscope's lamp on and off.

Some GRBL builds have "variable speed spindle" support: with these, the lamp brightness could be varied.

My use case is to reduce the amount of hardware needed for my microscope setup and tidy it up.
Running everything off the same 12V supply as the RAMPS board and motors would be very convenient, especially as the RAMPS boards (being 3D printer boards) have two high-current switch outputs (normally used for the extruder and bed heater).

This could be as simple as turning the lamp on when Pyuscope starts, and turning it off when it exits. Brightness control (if needed) could be added later.

JohnDMcMaster, 2024-02-10:
FYI there is a related test here: https://github.com/Labsmore/pyuscope-kitchen/blob/main/script/pick_place.py

JohnDMcMaster, 2024-02-10:
Related ticket: https://github.com/Labsmore/pyuscope/issues/322

JohnDMcMaster, 2024-02-23:
@philpem are you up to try this vs the new instrument subsystem plugin interface? Basically do this:
- Create a script PoC that can turn the lamp on / off
- Integrate that script as an instrument plugin

Sample that can actuate the spindle: https://github.com/Labsmore/pyuscope-kitchen/blob/main/script/pick_place.py

Once you test it, add an entry like this to your .pyuscope file:

```
    "instruments": {
        "illuminator": {
            "tab_name": "Illuminator",
            "visible": true,
            "plugin_path": "/home/mcmaster/doc/ext/pyuscope/uscope/script/test/instrument_simple.py",
        },
    },
```

Other: I'm not sure how graceful shutdown is if you close the whole window. But can test this out and improve it if its problematic

JohnDMcMaster, 2024-03-10:
I believe this is done per above, but haven't heard back. Closing ticket, let me know if you need more info

JohnDMcMaster, 2024-03-10:
Done using uscope/script/instrument/grbl_pwm.py

closed 2024-03-10 by JohnDMcMaster

philpem, 2024-03-11:
Hi John - sorry for the lack of response on this. I found out that my build of GRBL needs a little bit of work to turn on/off the outputs. I expect if you're sending the right G-code it'll just work. I need to find time to fix up GRBL but that's definitely a me-problem :D