Labsmore/pyuscope #449: wayland run note

JohnDMcMaster, 2024-02-23:
From will getting running on rpi5:

```
env -u WAYLAND_DISPLAY ./app/argus.py
```

This makes it work with X

JohnDMcMaster, 2024-02-23:
```
Pyuscope running on raspberry pi 5 latest raspberry pi OS Debian bookworm all updates installed, using the default wayland environment 

Changes made:
0. Had to reinstall git
1. Commented out the touptek function
2. Added --break-system-packages to all pip3 install lines
3. Prefacing the Argus.Py with 

env -u WAYLAND_DISPLAY

seems to get it, and other apps, to fall back to XWayland

To switch raspberry pi OS overall to xorg from wayland

you do in a terminal sudo raspi-config then

“6 Advanced options” menu then

“A6 Wayland"
```

nubcore, 2024-06-02:
Tested notes above successfully on fresh Raspberry Pi 4 bookworm install. Script file used attached for reference (note file extension changed).

Note when using the `raspi-config` that X11 option needs to be the one activated, vs Wayfire/Wayland, and then not needing to use `env -u` option on launch.

Full command used to launch from pyuscope directory, after running setup script, X11 config, and reboot:
`./app/argus.py --microscope lip-va1`

[setup_pi_sh.txt](https://github.com/user-attachments/files/15527258/setup_pi_sh.txt)