Labsmore/pyuscope #459: setup deps

CarlFK, 2024-03-05:
./setup_ubuntu_20.04.sh: line 38: unzip: command not found

JohnDMcMaster, 2024-03-07:
Debian 6?

CarlFK, 2024-03-07:
Debian 12, and -lite which is similar to ubuntu-server - a base that kinda only has enough stuff to apt install from the repos.

I have a few thoughts...

1. abandon this 'thing' and put efforts into a browser based UI. 

2. create a launchpad ppa, properly package 'this' and upload a .deb.  (yeah, packaging things is daunting.  I've never done it from scratch. )  I'm fishing for someone to offer guidance.  long shot they just do it.

3. start with the light os, (both ubuntu and debian) grind though the process of whack-a-mole and keep adding stuff to the current script, like add unzip to the apt install  https://github.com/Labsmore/pyuscope/blob/main/setup_ubuntu_20.04.sh#L32

JohnDMcMaster, 2024-03-08:
1: we are working on this already, but will still need a way to package / setup the embedded OS. So this mitigates the problem a bit but doesn't make it go away

2: Not out of the question, but we'd need to understand for example how a .deb can install flatpaks. This could just mean we need to create additional packages to make it work

3: no we are trying to avoid this. We'd rather put that effort into appimage, conda, etc. In the meantime though feel free to make suggestions and/or pull request