Labsmore/pyuscope #461: -ltoupcam error during install

zixb, 2024-03-08:
I'm getting the following error when running PYUSCOPE_MICROSCOPE=none ./setup_ubuntu_20.04.sh:

libtool: link:  gcc -shared  -fPIC -DPIC  .libs/libgsttoupcamsrc_la-gsttoupcamsrc.o .libs/libgsttoupcamsrc_la-gstplugin.o   -lgstbase-1.0 -lgstcontroller-1.0 -lgstreamer-1.0 -lgobject-2.0 -lglib-2.0 -ltoupcam -lgstvideo-1.0  -pthread -g -O2   -pthread -Wl,-soname -Wl,libgsttoupcamsrc.so -Wl,-version-script -Wl,.libs/libgsttoupcamsrc.ver -o .libs/libgsttoupcamsrc.so
/usr/bin/ld: cannot find -ltoupcam
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:434: libgsttoupcamsrc.la] Error 1

Not sure what to do at this point...

zixb, 2024-03-08:
Hmm... removing $(TOUPCAM_LIBS) from the following line in pyuscope\gst-plugin-toupcam\src\Makefile.am allowed it to proceed:

libgsttoupcamsrc_la_LIBADD = $(GST_LIBS) $(TOUPCAM_LIBS) -lgstvideo-1.0

BTW - I am running this in WSL in case that matters.

JohnDMcMaster, 2024-03-10:
Hi zixb! We haven't tested WSL, but curious what you find. To clarify, which Linux distribution under WSL are you using?

Can you give me a bit more background on what you are trying to do? Generally curious, but also you might not need libtoupcam depending on your application.

zixb, 2024-03-10:
Hi John,

The Linux distro is Ubuntu 20.04.6 LTS (GNU/Linux 5.15.146.1-microsoft-standard-WSL2 x86_64).  I'm going to try a VM to see if it installs any easier.  

As for what I'm doing, I was trying to see if I could get pyuscope working with my hardware .  I have an Amscope MU1803 camera and some other hardware I am cobbling together.  This is just a hobby for me so I can't really justify a Labsmore system and I am trying to do it on the cheap.  Sorry!  :)  Thanks for making this available, and if I find anything interesting I will let you know.

JohnDMcMaster, 2024-03-10:
All good! Just keep in mind the amount of support we can provide might be limited. If you run into issues, for example, probably better to stick to Ubuntu 20.04 since its the most well tested.

You might also be interested in the FOSS Microscopy discord: https://discord.gg/K2Psmyp6