Labsmore/pyuscope #473: Exposure & Color Temperature control on LIP-VA1

nubcore, 2024-06-02:
The "Imager" tab doesn't allow changing Exposure and doesn't give any options for Color Temperature. I know this isn't a bug per-say, but is there a work around that can be documented for editing this in a config or manually in some way?

nubcore, 2024-06-03:
Found suitable process for overriding and configuring most settings and getting generally improved results as baseline.

List all devices:
`v4l2-ctl --list-devices`

Set device to configure:
`v4l2-ctl -d /dev/video0`

List all available user controls, defaults and current values:
`v4l2-ctl -l`

Turn off automatic white balance:
`v4l2-ctl -c white_balance_automatic=0`

Set white balance close to cold white LED illuminator:
`v4l2-ctl -c white_balance_temperature=6500`

Set brightness:
`v4l2-ctl -c brightness=-42`

Set exposure to manual:
`v4l2-ctl -c auto_exposure=1`

Set esposure:
`v4l2-ctl -c exposure_time_absolute=1`

Some tips for getting started:
https://medium.com/@deepeshdeepakdd2/v4l-a-complete-practical-tutorial-c520f097b590

JohnDMcMaster, 2024-06-03:
I'll answer in a bit more detail later, but these settings were intentionally disabled due to various issues. You can re-activate them using pyuscope dev mode though if you want to play with them from the GUI itself.