Labsmore/pyuscope #475: Documentation request: how to derive the optical factors in the microscope config

philpem, 2024-06-16:
In the configuration for a microscope there are several parameters related to its optical setup. These seem to be fed into a bunch of places in Pyuscope. It would be good to have some documentation about which of these parameters are important, and how to derive them.

Reason being, I have an Olympus BH2 series Metallurgical microscope which I've customised by adding a Raspberry Pi HQ Camera and a motorised X-Y stage for use with Pyuscope. I also have objectives which aren't included in the default Pyuscope setup.

Settings like `native_pixel_pitch_um` seem fairly straightforward and I've got a working value from the IMX477 datasheet. However I have no idea how to pass details of the optical path to Pyuscope (there's a projection lens on the trinocular port, and a relay lens in the C-mount adapter), or create profiles for the objectives I have (Olympus MSPlan and Plan series).

The LIP-A1 config includes some configurations which seem to relate to a relay lens, but there are no details about how e.g, the `x_view` value was derived.

I have an Olympus OBMM reflective scale so can put that on the stage and take measurements if needed.

philpem, 2024-07-01:
I've been looking into this.

At the moment, I have `native_pixel_pitch_um = 1.55` in the config, which matches the raw pixel size of the IMX477 sensor. I'm finding that the overlap Pyuscope gives on scans is excessive. I could delete every other image and still be able to stitch them together.

On the optical side, my BH2 has a WF10x eyepiece mated with a set of objectives on the turret. If I had the 5x selected (as an example), the total magnification would be 5x times the 10x of the viewfinder, or 50x.

The camera setup on the BH2 is intended to match the field of view of the optical viewfinder, provided that the correct parts are used. With a 35mm film or digital SLR setup, there'd be a Photomicro Adapter and NFK 2.5x LD projection eyepiece on the trinocular turret. This should, in theory, produce an image with the same field of view as what would be seen through the microscope eyepiece. This makes a sort of sense and fits for most applications where the size of the object on the stage doesn't need to be known.

With a video camera mounted, the setup is slightly different. The same NFK 2.5x LD is in play, along with a 0.3x relay lens in the C-mount adapter, which steps the 43.26mm diagonal image (35mm film frame) down to 16mm diagonal for a 1" C-mount sensor. This gives mostly the same field of view (in theory). This would also apply to the IMX477 if it were a 1" sensor: but it's not, it's a 1/2.3" (7.857mm diagonal) sensor. That means there's a 16/7.856 ~= 2.04x magnification caused by the sensor being smaller than the projected image circle (aka "crop factor").

Pulling this discussion back to Pyuscope: from its perspective, during a pan scan, the X and Y motion step need to be calculated from several known factors:
  * The sensor's pixel size (1.55 um) and the X and Y pixel size of the image (4056 x 3040 pixels), giving the sensor's active area size
  * The magnification of the objective (say, 30%)
  * An optical scaling factor, which we need to figure out and apply to the pixel size.

The X and Y pixel size multiplied by the pixel size gives the active area size:
  * X: 4056 * 1.55e-6 = 6.2868mm
  * Y: 3040 * 1.55e-6 = 4.7182mm
  * Diagonal = `sqrt(6.2868^2 + 4.7182^2)` ~= 7.86 mm

To figure out the total magnification factor, I picked the 10x objective, found two features on the IC, and measured the stage X travel when they were positioned on the sides of the image field:
  * X: -7.033 - -8.673 = 1.64 mm
  * Y: 3.901 - 2.711 = 1.19 mm
  * Diagonal = `sqrt(1.64^2 + 1.19^2)` ~= 2.02 mm

Working backwards:
  * The sensor active area is 7.86mm diagonal
  * There is a 10x magnification caused by the objective: 7.86 / 10 = 0.786 mm
  * We needed to move 2.02 mm, so our scaling factor is 2.02 mm / 0.786 mm ~= 2.57

Unfortunately I haven't been able to work out how the lens factors (2.5x projection eyepiece, 0.3x relay lens) and crop factor (2.04x) end up in this number - but it seems to be correct because it gives accurate images with about 30% overlap, which matches the Pyuscope logs.

JohnDMcMaster, 2026-02-25:
For what its worth there is a sort of auto calibration suite that isn't public. I plan on releasing it / polishing it up once I get back to development in the near future